OxWearables/biobankAccelerometerAnalysis #6: gui.py not running in Linux

aidendoherty, 2016-01-29:
cmd > python gui.py 

output:
Traceback (most recent call last):
  File "gui.py", line 485, in <module> TkinterGUI(root).pack()
  File "gui.py", line 153, in __init__ self.advancedOptionInputs.append({'frame':frame, 'pack_opts': frame.pack_info()})
  File "/usr/lib/python2.7/lib-tk/Tkinter.py", line 1900, in pack_info self.tk.call('pack', 'info', self._w))
TypeError: coercing to Unicode: need string or buffer, _tkinter.Tcl_Obj found

aidendoherty, 2016-01-29:
My linux version of python is: Python 2.7.6

supersational, 2016-01-29:
Which linux distro is that on?

aidendoherty, 2016-01-29:
Lubuntu/Ubuntu 14.04.3 LTS

supersational, 2016-02-01:
[image]
Works for me with a fresh install of Ubuntu/Anaconda (2.7.6), are you using the standard python installation?

aidendoherty, 2018-12-07:
We no longer actively support GUI's.

closed 2018-12-07 by aidendoherty