OxWearables/biobankAccelerometerAnalysis #7: Possible to tidy visual appearance of collections.OrderedDict() in ActivitySummary.py?

aidendoherty, 2016-01-29:
Could the [block of code](https://github.com/computationalEpidemiology/biobankAccelerometerAnalysis/blob/master/ActivitySummary.py#L292) that starts with this:
```python
# store variables to dictionary
    result = collections.OrderedDict()
    result['file-name'] = rawFile
    f = '%Y-%m-%d %H:%M:%S'
    result['file-startTime'] = startTime.strftime(f)
    result['file-endTime'] = endTime.strftime(f)
...
```

Perhaps be styled to look like [this code block](https://github.com/computationalEpidemiology/biobankAccelerometerAnalysis/blob/master/gui.py#L25):
```python
self.target_opts = {
            'filename': Tkinter.StringVar(), # we use stringVar so we can monitor for changes
            'dirname': Tkinter.StringVar(),
            'filenames': [], # for multiple file selections, unused due to dodgy Tkinter support
            'target_type': Tkinter.StringVar(),
            'file_list': [],
            'file_opts': {
                'filetypes': [('all files', '.*'), ('CWA files', '.cwa')],
                'parent':root,
                'title': 'Select a file to process'
            },
            'dir_opts': {
                'mustexist': False,
                'parent': root,
                'title': 'Select a folder to process'
            }
        }
```

supersational, 2016-01-29:
We can do it, but it gets a bit messy when mixing with for loops : 
```
    result = {
        'file-name' : args.rawFile,
        'file-startTime': startTime.strftime('%Y-%m-%d %H:%M:%S'),
        'file-endTime': endTime.strftime('%Y-%m-%d %H:%M:%S'),
        # physical activity output variable (mg)
        'acc-overall-avg(mg)': formatNum(accAvg*1000, 2),
        'acc-overall-std(mg)': formatNum(accStd*1000, 2),
        # data integrity outputs
        'quality-goodWearTime': goodWearTime,
        'quality-goodCalibration': goodCalibration,
        'quality-calibratedOnOwnData': calibratedOnOwnData,
        'quality-daylightSavingsCrossover': daylightSavingsCrossover,
        # physical activity variation by day / hour
        'acc-mon-avg(mg)': formatNum(accDays[0]*1000, 2),
        'acc-tue-avg(mg)': formatNum(accDays[1]*1000, 2),
        'acc-wed-avg(mg)': formatNum(accDays[2]*1000, 2),
        'acc-thur-avg(mg)': formatNum(accDays[3]*1000, 2),
        'acc-fri-avg(mg)': formatNum(accDays[4]*1000, 2),
        'acc-sat-avg(mg)': formatNum(accDays[5]*1000, 2),
        'acc-sun-avg(mg)': formatNum(accDays[6]*1000, 2),
        'file-firstDay(0=mon,6=sun)': startTime.weekday()
    }
    for i in range(0, 24):
        result['acc-hourOfDay' + str(i) + '-avg(mg)'] = formatNum(accHours[i]*1000, 2)
    # wear time characteristics
    result = result.update({
        'wearTime-overall(days)': formatNum(wearTimeMins/1440.0, 2),
        'nonWearTime-overall(days)': formatNum(nonWearTimeMins/1440.0, 2),
        'wearTime-mon(hrs)': formatNum(wearDay[0]/60.0, 2),
        'wearTime-tue(hrs)': formatNum(wearDay[1]/60.0, 2),
        'wearTime-wed(hrs)': formatNum(wearDay[2]/60.0, 2),
        'wearTime-thur(hrs)': formatNum(wearDay[3]/60.0, 2),
        'wearTime-fri(hrs)': formatNum(wearDay[4]/60.0, 2),
        'wearTime-sat(hrs)': formatNum(wearDay[5]/60.0, 2),
        'wearTime-sun(hrs)': formatNum(wearDay[6]/60.0, 2)
    })
    for i in range(0,24):
        result['wearTime-hourOfDay' + str(i) + '-(hrs)'] = formatNum(wear24[i]/60.0, 2)
```

aidendoherty, 2016-01-30:
On further reflection, we will stick with collections.OrderedDict()  as this provides a consistent output. While it is visually appealing, unfortunately this potential code below results in unordered elements being stored to a dict and thus doesn't produce a nice structured output .json file.

```python
result = {
         'file-name' : args.rawFile,
         'file-startTime': startTime.strftime('%Y-%m-%d %H:%M:%S'),
         'file-endTime': endTime.strftime('%Y-%m-%d %H:%M:%S'),
         # physical activity output variable (mg)
         'acc-overall-avg(mg)': formatNum(accAvg*1000, 2),
         'acc-overall-std(mg)': formatNum(accStd*1000, 2),
 }
```

closed 2016-01-31 by aidendoherty