OxWearables/biobankAccelerometerAnalysis #11: Add option to specify nonWear thresholds

aidendoherty, 2016-06-13:
-nonWearThreshold to adjust cutoff point for nonWear time (currently 60mins)
 + some option to specify stdev threshold

supersational, 2016-06-13:
-nonWearThreshold is done, but just to note that in the OutputSummary.json we have changed : 
```
    "wearTime-numNonWearEpisodes(>1hr)": 62, 
```
To:
```
    "wearTime-nonWearThreshold(minutes)": 60, 
    "wearTime-numNonWearEpisodes": 1, 
```

(which may break some compatibility with old processing scripts)

supersational, 2016-06-13:
https://github.com/activityMonitoring/biobankAccelerometerAnalysis/commit/7fc73ab0a6ed6d2ea477ba0bc16126dfed699e6e

supersational, 2016-06-13:
The nonWearThreshold variable was originally called 'minDuration' in the code, would it be ok to change the parameter name to '-nonWearMinDuration' to match the original code naming?

Then we could call the standard deviation parameter '-nonWearStdThreshold' without having too many similarly named parameters.

supersational, 2016-06-13:
Second threshold option done here: https://github.com/activityMonitoring/biobankAccelerometerAnalysis/commit/110c118e936a28af65d888e0877e4f388512124d

aidendoherty, 2016-06-13:
Yes,  sounds much better

Best wishes,
Aiden

closed 2018-12-07 by aidendoherty