OxWearables/biobankAccelerometerAnalysis #12: Add option to disable data imputation

aidendoherty, 2016-06-13:
-diasbleImputation option to skip imputation (for the outputsummary.json)

supersational, 2016-06-13:
This can now be done by setting '-nonWearThreshold -1' in this commit: https://github.com/activityMonitoring/biobankAccelerometerAnalysis/commit/110c118e936a28af65d888e0877e4f388512124d 

However there are still some 'imputed' values in the AccTimeSeries file, which may be confusing to users (I think these are actually due to NaN values from the java process)

closed 2020-03-21 by chanshing

JianWang2016, 2020-04-14:
'-nonWearThreshold -1' doesn't seem to be merged in the master branch.

aidendoherty, 2020-04-15:
@JianWang2016 thanks for raising this. Could you perhaps create an issue to request this?

The line of code to toggle is https://github.com/activityMonitoring/biobankAccelerometerAnalysis/blob/master/accelerometer/summariseEpoch.py#L432

I'm unsure if this option should be provided, given the danger of interpreting non-imputed data. However, we could include a toggle with a suitable cautious/danger-laden name to highlight that it should be used with care.