OxWearables/biobankAccelerometerAnalysis #13: gui.py fails when the folder length is too long

supersational, 2016-12-15:
related? https://msdn.microsoft.com/en-us/library/aa365247(VS.85).aspx#maximum_path_length

TZebin, 2018-08-23:
Hi, 
I am updating the code to python 3.6.  For the gui.py, the GUI pops up okay(screenshot attached) when I execute it, but when I am choosing raw/bulk UKbiobank accelerometer file and press 'start', the following error appears while reading lines with the 'pollStdout' function: Any idea what's causing this error?

filename 8
python -u ActivitySummary.py C:/Batch_2/1000274_90001_0_0.cwa
running:  python -u ActivitySummary.py C:/Batch_2/1000274_90001_0_0.cwa
b"processing file 0 'C:/Batch_2/1000274_90001_0_0.cwa' with these arguments:\r\n"
Exception in thread Thread-6:
Traceback (most recent call last):
  File "C:\Users\mchijtz4\AppData\Local\Continuum\anaconda3\lib\threading.py", line 916, in _bootstrap_inner
    self.run()
  File "C:\Users\mchijtz4\AppData\Local\Continuum\anaconda3\lib\threading.py", line 864, in run
    self._target(*self._args, **self._kwargs)
  File "C:/Work/code/gui.py", line 474, in pollStdout
    self.textbox.insert(Tk.END, "\n" + line.rstrip())
TypeError: must be str, not bytes
[image]

aidendoherty, 2018-12-07:
Hello, thanks for highlighting this. Unfortunately we can no longer actively support the GUI code in this project. Very happy for you to have a go at updating this, particularly as our new update now relies on python3 (rather than previous python2 dependency).

closed 2018-12-07 by aidendoherty