OxWearables/biobankAccelerometerAnalysis #17: Error reading staticBoutsFile (Python 3.7, accelerometer/device.py)

NarayanSchuetz, 2019-04-18:
For some reason there occurs an index out of range error when it tries to read the static bouts file to calculate the calibration coefficients:

The problem occurs at the following line:

https://github.com/activityMonitoring/biobankAccelerometerAnalysis/blob/b5809e7eea1820a4477303751389f71c406be4c9/accelerometer/device.py#L210

Letting `numpy` read the file by itself solves the issue:

 ```
np.loadtxt(staticBoutsFile,delimiter=",",skiprows=1,
                usecols=(2,3,4,11,13))
```

Is there a specific reason why you are reading in the file in binary mode first that could alter the results?
As an additional note, the stationary file looked ok and didn't seem corrupted or anything. Reading it with `pandas` worked perfectly fine, as did reading it by `numpy` directly.

chanshing, 2020-03-21:
Your solution is probably OK. I am not sure why the error happened, or whether it was related to https://github.com/activityMonitoring/biobankAccelerometerAnalysis/commit/9f6e6360f03fdea705b4418eb1dc286d461dc1e3

aidendoherty, 2020-05-04:
This code has now been made redundant and the stationary file is now read by pandas.

closed 2020-05-04 by aidendoherty