OxWearables/biobankAccelerometerAnalysis #18: KeyError: "filename 'featureCols.txt' not found"

Kelpies1, 2019-05-31:
I was processing a sample.cwa file and got the following error message.  Is there any suggestion that you can provide us with based on the traceback message.

Traceback (most recent call last):
  File "accProcess.py", line 344, in <module>
    main()  # Standard boilerplate to call the main() function to begin the program.
  File "accProcess.py", line 270, in main
    verbose = args.verbose)
  File "/home/kelpie/biobankAccelerometerAnalysis/accelerometer/summariseEpoch.py", line 103, in getActivitySummary
    e, labels = accClassification.activityClassification(e, activityModel)
  File "/home/kelpie/biobankAccelerometerAnalysis/accelerometer/accClassification.py", line 37, in activityClassification
    featureCols = getListFromTxtFile(getFileFromTar(activityModel, 'featureCols.txt'))
  File "/home/kelpie/biobankAccelerometerAnalysis/accelerometer/accClassification.py", line 407, in getFileFromTar
    array_file.write(t.extractfile(targetFile).read())
  File "/home/kelpie/anaconda3/lib/python3.7/tarfile.py", line 2068, in extractfile
    tarinfo = self.getmember(member)
  File "/home/kelpie/anaconda3/lib/python3.7/tarfile.py", line 1754, in getmember
    raise KeyError("filename %r not found" % name)
KeyError: "filename 'featureCols.txt' not found"

chanshing, 2019-05-31:
In future versions, the features of the classifier will be provided by the classifier model (and should come in the tar file). In the meantime you can revert to a previous version of `accelerometer/accClassification.py` that contains the hard-coded features. Use the following:

`git checkout 25ef9649c21c983fa2db372df830ddd9270abaed accelerometer/accClassification.py`

aidendoherty, 2019-05-31:
Thanks Shing! I introduced a bug, and will have this updated and fixed on Monday when I return to the office.