OxWearables/biobankAccelerometerAnalysis #19: Minor error messages and warnings

MustafaOguz, 2019-06-03:
I encountered the following error messages when processing accelerometer files. Even though they do not have an impact on the analyses, I thought I would note them here: 

1) Device.py:250: RuntimeWarning: invalid value encountered in double_scalars improvement = (bestError-rms)/bestError

2)	Java.time.DateTimeException:Invalid value for MonthfYear (valid values 1 – 12): 0 

3)	SettingwithCopyWarning: A value is trying to be set on a copy of a slice from a DataFrame. Try using .loc[row_indexer, col_indexer] = value instead

chanshing, 2019-06-22:
Thanks for raising these issues. The `RuntimeWarning: invalid value encountered in double_scalars improvement = (bestError-rms)/bestError` should now be fixed. It was due to the use of `np.inf` to indicate a very large number, which resulted in `np.nan` values in the first estimate of `improvement` during the iteration, although this did not affect the outcome of the run.
We will be looking into the remaining issues.

aidendoherty, 2020-05-04:
I think these issues are now gone, so will close this issue. @MustafaOguz please do let us know if this problem still persists for you.

closed 2020-05-04 by aidendoherty