OxWearables/biobankAccelerometerAnalysis #20: Can't import accelerometer No module named 'numpy.testing.decorators'

muschellij2, 2020-03-20:
Installed via the REAMDE:
```
MacBook-Pro-9:biobankAccelerometerAnalysis johnmuschelli$ python3
Python 3.5.1 (v3.5.1:37a07cee5969, Dec  5 2015, 21:12:44) 
[GCC 4.2.1 (Apple Inc. build 5666) (dot 3)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import accelerometer
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/Users/johnmuschelli/Dropbox/Packages/biobankAccelerometerAnalysis/accelerometer/__init__.py", line 1, in <module>
    import accelerometer.accClassification
  File "/Users/johnmuschelli/Dropbox/Packages/biobankAccelerometerAnalysis/accelerometer/accClassification.py", line 6, in <module>
    import pandas as pd
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/site-packages/pandas/__init__.py", line 56, in <module>
    import pandas.util.testing
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/site-packages/pandas/util/testing.py", line 22, in <module>
    from numpy.testing.decorators import slow     # noqa
ImportError: No module named 'numpy.testing.decorators'
```
May be related to:  https://github.com/apache/incubator-mxnet/issues/17154

chanshing, 2020-03-21:
It seems to be related to this:
https://stackoverflow.com/questions/59474533/modulenotfounderror-no-module-named-numpy-testing-nosetester
Please can you show your `scipy` and `numpy` versions? Try upgrading `scipy`.

muschellij2, 2020-03-21:
Still failing after updating `scipy`:
```
MacBook-Pro-9:biobankAccelerometerAnalysis johnmuschelli$ pip freeze | grep numpy
numpy==1.18.1
MacBook-Pro-9:biobankAccelerometerAnalysis johnmuschelli$ pip freeze | grep scipy
scipy==1.1.0
```

chanshing, 2020-03-21:
Can you try downgrading `numpy` to `1.18`

muschellij2, 2020-03-22:
Downgraded, same thing.

chanshing, 2020-03-23:
Can you try upgrading `pandas`? Or create a fresh python environment and install `pandas` first.

angerhang, 2020-04-20:
BTW @muschellij2 you should also use `Python 3.7` instead of `Python 3.5.1.`.

aidendoherty, 2020-05-04:
I think this issue is now solved following suggestion of @angerhang . @muschellij2 please reopen this if you still have problems.

closed 2020-05-04 by aidendoherty