OxWearables/biobankAccelerometerAnalysis #53: Interpretation of circadian rhythm analysis output

JianWang2016, 2020-06-10:
@Alaina121, @R-Walmsley, and @chanshing,

Thanks much for circadian rhythm analysis, could you please describe how to interpret the output?

R-Walmsley, 2020-06-12:
Hi @JianWang2016

There are three different outputs.

- M10L5 refers to Relative Amplitude, a widely used measure of circadian rhythmicity (e.g. https://pubmed.ncbi.nlm.nih.gov/29776774/). For each 10am-10am 24 hour period, the most active continuous 10 hour period and the least active continuous 5 hour period are found. The average acceleration during these periods is calculated, and averaged over the days of wear (to give a value M10 for the average acceleration in the most active 10 hour periods and a value L5 for the average acceleration in the least active 5 hour periods). Then the Relative Amplitude is returned as (M10-L5)/(M10+L5).

- PSD (Power Spectral Density) and Fourier Frequency are more novel measures derived as part of this work. These are still being developed and explored, so there may be further developments. If fourierWithAcc = False (the default), they use the sleep/wake signal and perform a discrete Fourier transform on this. The PSD value returned is derived from the Power Spectral Density for the 24 hour period (see for example https://ccrma.stanford.edu/~jos/mdft/ and the section on Spectral Bin Numbers- although we normalise slightly differently). We interpret higher values as meaning more dominant 24 hour rhythmicity in the sleep-wake signal, lower values as meaning more other frequencies are present, indicating more irregular or broken sleep. fourierFrequency identifies the frequencies with the highest Power Spectral Density i.e. the most prominent frequencies in the data (we haven't found this that useful, though). If fourierWithAcc = True, both these calculations use the acceleration signal rather than the sleep-wake signal - but we've found this measure to be a bit unstable and hard to interpret.

I hope that helps!! @Alaina121, @chanshing or @aidendoherty may be able to add/correct what I've said. Do get back in touch if you have further questions or anything about the behaviour doesn't make sense!

JianWang2016, 2020-06-16:
Thanks much @R-Walmsley.  This is very helpful.  When imputation is turned off, how would M10L5 handle the missing data? Would it skip the 24 hour day entirely? Imputation wouldn't make sense if many hours of a day are non-wear.

R-Walmsley, 2020-06-17:
Hi @JianWang2016, 
These functions currently use the imputed acceleration data regardless of whether imputation is switched on or off with the new --useRecommendedImputation flag.
This is something we could change in future. However, I think it would need thinking and discussion about what makes sense in different use cases: would we only include days with no non-wear time? This is not ideal as we would then drop a lot of days.

JianWang2016, 2020-06-25:
Much thanks!

closed 2020-06-25 by JianWang2016