OxWearables/biobankAccelerometerAnalysis #58: csv input file error

RachelOwl, 2020-07-01:
Hi 
I am trying run  python accProcess.py file.csv
csv has X ,Y , Z and time in this format: "2020-05-06 11:08:41.930"
I keep getting this message:
"Must specify either 'csvStartTime' (YYYY-MM-DDThh:mm format) and 'csvSampleRate', or:
'csvXYZYCols' (at least 4 values) and 'csvTimeFormat'
options for .csv type file: data/watch.csv
['java', '-classpath', 'java:java/JTransforms-3.1-with-dependencies.jar', '-XX:ParallelGCThreads=1', 'AccelerometerParser', 'data/watch.csv', 'timeZone:Europe/London', 'timeShift:0', 'outputFile:data/watch_-stationaryPoints.csv.gz', 'verbose:False', 'filter:True', 'getStationaryBouts:true', 'epochPeriod:10', 'stationaryStd:0.013', 'sampleRate:100']
Error: java calibration failed, exit  255"

but in the accelerometerParser,java  all variables are there:
I changed these vars as following:
`// Must supply additional information when loading from a .csv file
   LocalDateTime csvStartTime = "2020-05-06 11:08:41.930"; // start date of first sample
   double csvSampleRate = 25; // must specify sample rate if time column and date format not given
   int csvStartRow = 1;`

can you explain why the csv can't calibrate?

muschellij2, 2023-09-19:
Was this ever resolved?

chanshing, 2023-09-20:
@muschellij2 It should be. Are you facing issues processing a CSV file?

closed 2023-10-24 by chanshing