OxWearables/biobankAccelerometerAnalysis #71: tz-naive and tz-aware date time comparison

angerhang, 2020-10-06:
While I was using the `--endTime 2017-04-18T18:00` in a format that is compliant with the documentation https://github.com/activityMonitoring/biobankAccelerometerAnalysis/blob/0efaf0267d81121b0e244979c3efc1635d7fc8c2/accProcess.py#L51-L52 I ran to a type error 
```python
TypeError: Cannot compare tz-naive and tz-aware datetime-like objects
```
as it was trying to compare two timezone-naive and timezone-aware objects together:

This was caused by https://github.com/activityMonitoring/biobankAccelerometerAnalysis/blob/4ffefac5861a1ec58c28a8fe2bc3bee249cb93e2/accelerometer/summariseEpoch.py#L92-L93

as `e.index` is tz-aware but `endTime` is not. Probably we will have the same issue, when we supply the start time option.

I think https://github.com/activityMonitoring/biobankAccelerometerAnalysis/commit/379ebb2025f2ed44334c631288b85b2f3e660e6b  introduced the error as our output became time zone aware. 

There are two ways that we could resolve this issue, one is to change the `endTime` and `startTime` type and make them tz-aware. Another way is to convert `e.index` to be tz-naive for comparison. But the former approach is probably better. 

@chanshing what do you think, since you worked on this piece of code? :D

chanshing, 2020-10-07:
Thanks for identifying this @angerhang 
I think the first approach will involve less work, i.e. to localize the `endTime`.
You could try to momentarily convert the index to tz-naive but then you'd need to make sure everything is right.

angerhang, 2020-11-09:
Fixed in pr #74.

closed 2020-11-09 by angerhang