OxWearables/biobankAccelerometerAnalysis #75: Check gt3x v1 and v2 timestamp differences

angerhang, 2020-10-13:
Thanks @chanshing  for bringing this up.

> Just to double check, is this line correct?:
>https://github.com/activityMonitoring/biobankAccelerometerAnalysis/blame/6021de453950e464737373cd62ea7c6a6af3a391/java/ActigraphReader.java#L479
> Also, I see that in some parts firstSampleTime is multplied by 1000 to get seconds:
>https://github.com/activityMonitoring/biobankAccelerometerAnalysis/blame/6021de453950e464737373cd62ea7c6a6af3a391/java/ActigraphReader.java#L525
> but in other places it's not:
>https://github.com/activityMonitoring/biobankAccelerometerAnalysis/blame/6021de453950e464737373cd62ea7c6a6af3a391/java/ActigraphReader.java#L380
>Are these correct?

angerhang, 2020-10-13:
We should check this before PR https://github.com/activityMonitoring/biobankAccelerometerAnalysis/pull/70 is merged!

chanshing, 2023-03-01:
I believe this is fixed. Re-open otherwise.

closed 2023-03-01 by chanshing