OxWearables/biobankAccelerometerAnalysis #102: sklearn.ensemble.forest changed to sklean.ensemble._forest

xiangnandang, 2021-01-11:
https://github.com/activityMonitoring/biobankAccelerometerAnalysis/blob/61b36d1ae6212de93ec10a3cb164d40b2f6b5ace/accelerometer/accClassification.py#L9

It appears that since scikit-learn 0.22 (didn't figure out exactly which version), sklearn.ensemble.forest has changed to sklearn.ensemble._forest

since the package only requires scikit-learn >= 0.21.2 (https://github.com/activityMonitoring/biobankAccelerometerAnalysis/blob/61b36d1ae6212de93ec10a3cb164d40b2f6b5ace/setup.py#L23)

This cold lead to package not able to run, even though most people wouldn't even use that part of the package to train model

xiangnandang, 2021-01-11:
sorry, just saw #101 , closing this one

closed 2021-01-11 by xiangnandang

angerhang, 2021-01-11:
Thanks for reporting nonetheless, the dev team would meet this week to resolve this issue. But for a quick fix, just install 0.22 sklean should fix it temporally. Sorry for the inconvenience.