OxWearables/biobankAccelerometerAnalysis #103: 5 second of epoch leads to all timeseries with imputed=1

xiangnandang, 2021-01-12:
https://github.com/activityMonitoring/biobankAccelerometerAnalysis/blob/61b36d1ae6212de93ec10a3cb164d40b2f6b5ace/accelerometer/accUtils.py#L476

if 5 second epoch is used, `avgArmAngelAbsDiff` from epoch file / df will always be `na`, thus `imputed` in timeseries file will always be `1`

This seemed to work before: `e['imputed'] = np.isnan(e['acc']).astype(int)`, but there might be reason I'm not aware to use current `e_new['imputed'] = e.isna().any(1).astype('int')`

angerhang, 2021-01-12:
Great question @xiangnandang. Thanks for reporting.

@chanshing this behavior is introduced in https://github.com/activityMonitoring/biobankAccelerometerAnalysis/commit/379ebb2025f2ed44334c631288b85b2f3e660e6b. For the imputed field, shouldn't the behavior be that if only `acc` is NA then, we should mark an epoch as imputed instead of the whole frame? Is this because if there is any features of an epoch that cannot be computed, we would have to impute that epoch?

chanshing, 2021-01-12:
> Great question @xiangnandang. Thanks for reporting.
> 
> @chanshing this behavior is introduced in [379ebb2](https://github.com/activityMonitoring/biobankAccelerometerAnalysis/commit/379ebb2025f2ed44334c631288b85b2f3e660e6b). For the imputed field, shouldn't the behavior be that if only `acc` is NA then, we should mark an epoch as imputed instead of the whole frame? Is this because if there is any features of an epoch that cannot be computed, we would have to impute that epoch?

Hmmm AFAIK we don't impute features other than the mean acceleration. But we do sort of impute the predicted activities -- we average the predictions for the other days for that same epoch. So if any feature was NaN, then the prediction would be NaN too for that epoch, then it would need to be imputed.

If I understand correctly, `avgArmAngelAbsDiff` is a new feature that is not used for activity classification, so `avgArmAngelAbsDiff` may be NaN but the classifier is still working. So `e_new['imputed']` should actually look at the features used by the classifier rather than the whole.

xiangnandang, 2021-01-12:
Thank you both for responding so quickly!
@chanshing , yes, your understanding is correct!

chanshing, 2021-01-13:
> Thank you both for responding so quickly!
> @chanshing , yes, your understanding is correct!

Thank you @xiangnandang 
In that case let's think about what's the best way to resolve this @angerhang ideas?

chanshing, 2021-01-15:
@angerhang It seems that it also doesn't work for exactly 5 secs no? https://github.com/activityMonitoring/biobankAccelerometerAnalysis/blob/cc9beb73dca1e118de232d195bd644b0af1fad3e/java/Features.java#L624

chanshing, 2021-01-15:
Just submitted a PR to fix this: https://github.com/activityMonitoring/biobankAccelerometerAnalysis/pull/105

@angerhang There's still an issue in `avgArmAngelAbsDiff` being NaN for exactly 5s-epoch. I'll leave that PR to you.

@xiangnandang Note that our ML model was trained with 30s-epoch so the predicted activities will be sort of garbage if you use something different. Here's the output sample plot for 5s-epoch:

[image]

closed 2021-01-28 by chanshing