OxWearables/biobankAccelerometerAnalysis #121: Meaning of summary['wearTime-diurnal*']

chanshing, 2021-02-16:
I'm a bit confused by the meaning of `summary['wearTime-diurnalHrs']` and `summary['wearTime-diurnalMins']`

https://github.com/activityMonitoring/biobankAccelerometerAnalysis/blob/cda85c70e1dff4c34ce311988f281336413ce3ee/accelerometer/summariseEpoch.py#L282-L285

In the code it seems to be computing the total number of hours and mins in a day.

aidendoherty, 2021-02-16:
Hi @chanshing the aim here is to capture the number of unique hours (max = 24) and minutes of data (max = 1440). This is very helpful to investigate if participants have worn a device over a complete 24hr period to ensure we have captured their entire circadian profile.

On reflection the labels aren't great. Perhaps it would be better understood if we renamed these to: wearTime-uniqueHrs and wearTime-uniqueMins ?

chanshing, 2021-02-17:
Ah that makes sense. Thanks Aiden!

closed 2021-02-17 by chanshing