OxWearables/biobankAccelerometerAnalysis #146: Dependency discrepancy

angerhang, 2021-04-09:
Currently, we specify the dependency version at two places, which sometimes can conflict each other. It seems a bit odd to have this redundancy? I remember @R-Walmsley was working on the dependency management, any ideas of how we should resolve this?

https://github.com/activityMonitoring/biobankAccelerometerAnalysis/blob/33400a055ebae5bf268a636addd66c54adf094bd/setup.py#L23

https://github.com/activityMonitoring/biobankAccelerometerAnalysis/blob/33400a055ebae5bf268a636addd66c54adf094bd/requirements.txt#L24

I came to this when I ran `pip3 install --upgrade -r requirements.txt` per `README.md` which gave me an error of 
```
ERROR: pip's dependency resolver does not currently take into account all the packages that are installed. This behaviour is the source of the following dependency conflicts.
accelerometer 2.0 requires scikit-learn==0.22, but you have scikit-learn 0.24.1 which is incompatible.
```

I just checked our CI tests, I think it is using a different command to install ` python -m pip install --upgrade pip`. That's why CI is not catching the error but we should modify the CI test as well to reflect our final decision.

R-Walmsley, 2021-04-09:
I think `setup.py` should really now be legacy, and isn't included in the installation instructions (as I took installation out). The only reason I didn't delete it at the time of adding `requirements.txt` was not being sure if the change would be permanent.

However, yesterday we decided installation should go back in, and `setup.py` is needed when you run that workflow so probably we need to make `setup.py` and `requirements.txt` link somehow. 

One option is to move everything to `setup.py` (optionally with `requirements.txt` referencing that: https://stackoverflow.com/questions/14399534/reference-requirements-txt-for-the-install-requires-kwarg-in-setuptools-setup-py) 

Sorry about this, was me trying to improve the dependency management without fully understanding it!

angerhang, 2021-04-09:
> https://stackoverflow.com/questions/14399534/reference-requirements-txt-for-the-install-requires-kwarg-in-setuptools-setup-py

Hmm thanks for the quick reply. The reference is very helpful. Looks like that `setup.py` and `requirements.txt` serve very different purposes. The former is for development purpose and the latter is more for deployment. 

I think this is also needed for the PyPI/Conda distribution.

R-Walmsley, 2021-04-12:
Hmm, I largely agree. 

I think we need to get this sorted in both the short and long term:

- In the short term, I'm concerned that currently we're not providing any deployment-mode instructions and we need to provide this, ideally with minimal overhead possible. Maybe this is to move back to `setup.py`, move the tests to use `setup.py` and delete `requirements.txt` altogether? @chanshing @angerhang if you approve this I can do this by tomorrow, ideally one of you can test it somewhere and then we can merge it ASAP. 
- In the long term, we need to figure out a robust balance between deployment mode, development mode, versions,, dependencies etc without duplication and redundancy. Think this will take a bit longer to get something we're happy with, and don't think we can wait for that to fix the above.

angerhang, 2021-04-12:
https://stackoverflow.com/questions/43658870/requirements-txt-vs-setup-py gives us a good summary especially the 2nd most voted answer.

In the short-term, we can't delete `req.txt` as we will need this to install dependencies. We might have a long-term solution too by moving everything into `setup.py` and change the content of `req.txt` into an onliner. setup.py` is used for package install such as pip install/python install module-name if that makes sense.

chanshing, 2022-03-01:
I think this is resolved now since PyPI

closed 2022-03-01 by chanshing