OxWearables/biobankAccelerometerAnalysis #150: [Installation] Can't download data models

joanitachong, 2021-04-16:
During installation, in utilities/downloadDataModels.sh , It says "wget: command not found" in lines 20-25 and 30. 

Then I tried accessing the "downloadDir" website from where the files are downloaded and I don't have access to it. It says "Forbidden: You don't have permission to access /aidend/accModels/ on this server."

What can I do?

chanshing, 2021-04-16:
I would seem that you don't have `wget`. Are you using Windows? Mac?

joanitachong, 2021-04-16:
I have Mac

chanshing, 2021-04-16:
I think you'll need to get `wget` first. I found this https://www.jcchouinard.com/wget/#:~:text=download(url)%20filename-,Debug%20Wget%20Command%20Not%20Found,that%20you%20installed%20wget%20properly.
@angerhang Any ideas?

joanitachong, 2021-04-17:
I just got wget, but I'm still having the same problem. I can't access the "downloadDir" ( http://gas.ndph.ox.ac.uk/aidend/accModels/ ). I'm still "Forbidden: You don't have permission to access /aidend/accModels/ on this server."

chanshing, 2021-04-17:
Are you still not able to run `bash utilities/downloadDataModels.sh`?

joanitachong, 2021-04-17:
It runs, but it can't find the download directory.

[image]

joanitachong, 2021-04-17:
I mean, the first 5 lines with (No such file or directory) are normal because those files don't exist. But the remaining ones (Box 1) should work. But they don't work because I don't have access to the download directory (Box 2 and 3). 

[image]

[image]

chanshing, 2021-04-17:
I don't think the access is the problem because I also don't have it and can still download the files just fine. You can have no access to the directory while having access to particular files. From what you showed, it seems that the problem is still `wget`.
Alternatively, you can try from your browser:

http://gas.ndph.ox.ac.uk/aidend/accModels/doherty-may20.tar
http://gas.ndph.ox.ac.uk/aidend/accModels/willetts-may20.tar
http://gas.ndph.ox.ac.uk/aidend/accModels/walmsley-nov20.tar
http://gas.ndph.ox.ac.uk/aidend/accModels/doherty-jan21.tar
http://gas.ndph.ox.ac.uk/aidend/accModels/willetts-jan21.tar
http://gas.ndph.ox.ac.uk/aidend/accModels/walmsley-jan21.tar

joanitachong, 2021-04-17:
You are right. I was able to download the files from the browser, so the problem must be the `wget`. But I got the files, so thanks!

closed 2021-04-17 by chanshing