OxWearables/biobankAccelerometerAnalysis #169: java.lang.ClassNotFoundException when running in windows

dwijenchawra, 2021-07-12:
Hello, 

I recently tried running the biobank program on Windows instead of UNIX, and encountered this issue:

`Error: Could not find or load main class AccelerometerParser
Caused by: java.lang.ClassNotFoundException: AccelerometerParser`

I went through a fresh install following the guide in the documentation, ran the bash commands in unix and copied the files to the right folders, and installed all the pip packages. 

Here is the full error output
`2021-07-12 15:02:38     === Calibrating ===
Error: Could not find or load main class AccelerometerParser
Caused by: java.lang.ClassNotFoundException: AccelerometerParser
['java', '-classpath', 'java:java/JTransforms-3.1-with-dependencies.jar', '-XX:ParallelGCThreads=1', 'AccelerometerParser', 'C:\\Users\\--------\\Documents\\VirtualboxShared\\biobankAccelerometerAnalysis\\data\\sample.cwa.gz', 'timeZone:Europe/London', 'timeShift:0', 'outputFile:C:\\Users\\----------\\Documents\\VirtualboxShared\\biobankAccelerometerAnalysis\\data\\sample-stationaryPoints.csv.gz', 'verbose:False', 'filter:True', 'getStationaryBouts:true', 'epochPeriod:10', 'stationaryStd:0.013', 'sampleRate:100', "csvTimeFormat:yyyy-MM-dd HH:mm:ss.SSSxxxx '['VV']'", 'csvStartRow:1', 'csvTimeXYZColsIndex:0,1,2,3']
Error: java calibration failed, exit  1`

Any tips?

angerhang, 2021-07-13:
Hey that's expected. At the moment we don't support Windows yet. https://github.com/activityMonitoring/biobankAccelerometerAnalysis/pull/168/files gives you the option to adapt the path to the java class file to the one you want.

You will want to do something like this: https://github.com/angerhang/biobankAccelerometerAnalysis/commit/1f5f0507256721536c622a4479336f256e39a85d

Hope this helps.

chanshing, 2022-10-07:
Fixed in https://github.com/OxWearables/biobankAccelerometerAnalysis/releases/tag/v5.1.3

closed 2022-10-07 by chanshing