OxWearables/biobankAccelerometerAnalysis #170: `cwa_regression` fails due to missing model resource

angerhang, 2021-07-15:
In our CI tests, `cwa_regression` will fail from time to time due to the following error:
```
--2021-07-14 15:37:34--  https://gas.ndph.ox.ac.uk/aidend/accModels/walmsley-jan21.tar
5179
Resolving gas.ndph.ox.ac.uk (gas.ndph.ox.ac.uk)... 163.1.206.75
5180
Connecting to gas.ndph.ox.ac.uk (gas.ndph.ox.ac.uk)|163.1.206.75|:443... connected.
5181
Unable to establish SSL connection.
5182
URL transformed to HTTPS due to an HSTS policy
```   
It sometimes fails using python 3.6 , 3.7, 3.8. Pretty annoying not sure what is causing this but someone will need to investigate this. I have started the CI tests for some of the current PRs.

angerhang, 2021-07-15:
After I re-run the tests, the issue persists. When I am running the download script on my local machine, I also have the same issue.

```
Resolving gas.ndph.ox.ac.uk (gas.ndph.ox.ac.uk)... 163.1.206.75
Connecting to gas.ndph.ox.ac.uk (gas.ndph.ox.ac.uk)|163.1.206.75|:443... connected.
ERROR: cannot verify gas.ndph.ox.ac.uk's certificate, issued by ‘CN=QuoVadis Global SSL ICA G3,O=QuoVadis Limited,C=BM’:
  Unable to locally verify the issuer's authority.
To connect to gas.ndph.ox.ac.uk insecurely, use `--no-check-certificate'.
```
It appears to be related to the domain certificate as reported earlier in https://github.com/activityMonitoring/biobankAccelerometerAnalysis/issues/153. 

I also tried the download script on VM, the download time out on 
```
--2021-07-15 06:23:11--  http://gas.ndph.ox.ac.uk/aidend/accModels/willetts-may20.tar
Resolving gas.ndph.ox.ac.uk (gas.ndph.ox.ac.uk)... 163.1.206.75
Connecting to gas.ndph.ox.ac.uk (gas.ndph.ox.ac.uk)|163.1.206.75|:80... 

failed: Connection timed out.
Retrying.
```

Likely it is not an issue specific to Mac but making all OS unstable for resource download.

Can @chanshing  or @R-Walmsley reproduce the same issues?

angerhang, 2021-07-15:
@aidendoherty who did you suggest that we should contact to double check the domain certificate? Is it the NDPH IT?

aidendoherty, 2021-07-15:
Thanks @angerhang - I've just sent an email to NDPH IT to see if we can resolve this. It does seem like a server issue. As a backup, I wonder if there are other (preferably free, or very cheap) places we can host the models? (e.g. AWS?)

angerhang, 2021-07-19:
It is so decided that we will first try download the model without ``--no-check-certificate``. If it fails, then we will try ``--no-check-certificate`` flag.

A patch is coming up soon.

PR #172 (2021-07-21): Fall back for downloadDataModel
This should fix #170 allowing `` --no-check-certificate`` to be used when the certificate cannot be verified. 

All the CI tests should pass with this change.
closed 2021-07-21 by aidendoherty