OxWearables/biobankAccelerometerAnalysis #174: Overclassification of sleep for low frequency aquisitions (5Hz)

SuperCorks, 2021-09-03:
Hey guys, let me start off by saying thank you very much for this library! If I can tip you in anyway, please let me know how.

We're developing our own device for a study here and gathered a first dataset. The wearable doesn't have too much memory (15MB) and the motion sensor (accel/gyro) is the sensor that produces the most data (12 bytes per sample) so we limited it at 5Hz. Also the sensor can be configured with a precision/max value tradeoff so we went with a max value of 2g (other options are 4g or  8g). 

We're now running different actigraphy algorithms, including yours, on our data and seeing some weird results as shown below:
 
[image]
[image]
[image]

I understand our setup isn't ideal, so here are my questions:
*  It looks like the model over-identifies sleep. With our current 5Hz setup, is there something we can do to make the models more accurate?
*  What is the best configuration of our sensor that your suggest? Here are some possible examples:
    *  20 Hz with a max of 4g?
    *  32 Hz with a max of 4g?
    *  32 Hz with a max of 8g?

angerhang, 2021-09-04:
Thanks for your questions. It is always nice to have external validation using our algorithm. `5Hz` does seem a bit too low for sleep, making the device less sensitive to movement. To me 4/8g in terms of range doesn't make a big difference but something with 32Hz would likely help with your classification output.

@chanshing @R-Walmsley @aidendoherty do they also need to normalise the input for the random forest model by any chance?

chanshing, 2021-09-04:
5Hz is indeed way too low. For example, slow walking is at around 2Hz already (2 steps/sec), meaning you need _at least_ >4Hz to capture it (Nyquist criteria). Sprinting is almost 5Hz, so >10Hz there.
In terms of range, 4g is more than enough from my experience (like <0.1% of free-living data I've worked with exceeds 3g).
For reference, Fitbit and Apple Watch use something like >100Hz and +/-16g which I think might be too much.
I think +/-4g @ 50Hz should be enough. Given your options, I would go for the 32Hz, 4g

SuperCorks, 2021-09-07:
Alright thanks to all of you for you answers!

closed 2021-09-07 by SuperCorks