OxWearables/biobankAccelerometerAnalysis #181: Refactor plotting

angerhang, 2021-11-02:
At the moment, the plotting script is a big script clotting together which makes it more cumbersome than ideal to make changes or add features. We need to refactor this badly.

https://github.com/activityMonitoring/biobankAccelerometerAnalysis/blob/master/accelerometer/accPlot.py

chanshing, 2021-11-03:
I agree (y)

chanshing, 2022-10-07:
@angerhang Would you like to look at `accPlot` again and have a go?

R-Walmsley, 2022-10-07:
I know @gmertes is also using this function so cc'ing in case you've already made any changes to your own version.

gmertes, 2022-10-07:
My changes are already merged. My only request is to keep the interface to [plotTimeSeries](https://github.com/OxWearables/biobankAccelerometerAnalysis/blob/446295152d4750f4ae7daae6a5db83740bae9a3d/accelerometer/accPlot.py#L92) the same (or at least backwards compatible).

chanshing, 2023-03-01:
Re-open if still unfixed.

closed 2023-03-01 by chanshing