OxWearables/biobankAccelerometerAnalysis #186: Activity classification model performances and sampling rate

ghammad, 2021-12-01:
If I am not mistaken, the default sampling rate for the data used to develop and validate the model described in Willetts et al., 2018 and made available through the "biobankAcc" package, is 100Hz.

Does the impact of the sampling rate on the model performances have been studied? I.e. is it possible to use the model with data acquired at a lower sampling rate (ex. 50 Hz)?

If I understand correctly, features in a frequency domain up to 15Hz were used... so still below the Nyquist frequency for a sampling rate of 50Hz... so there is hope, isn't it?

Thank you.

chanshing, 2021-12-01:
Hi. Yes, you can use a lower sampling rate. I think our features are resolution-invariant so it should work.

ghammad, 2021-12-01:
Ok. Thank you.

closed 2021-12-01 by chanshing

aidendoherty, 2021-12-03:
Thanks for this question @ghammad 
This paper might also be of interest:

Small, S., Khalid, S., Dhiman, P., Chan, S., Jackson, D., Doherty, A., & Price, A. (2021). Impact of Reduced Sampling Rate on Accelerometer-Based Physical Activity Monitoring and Machine Learning Activity Classification, Journal for the Measurement of Physical Behaviour, 4(4), 298-310. Retrieved Dec 3, 2021, from https://journals.humankinetics.com/view/journals/jmpb/4/4/article-p298.xml

ghammad, 2021-12-06:
Hi @aidendoherty 

Thank you for the reference. I missed it! Very useful paper as, working in a chronobiology lab, from our perspective, the added value of actigraphy with respect to other techniques (among other advantages), resides in long recordings.

Cheers,

Greg