OxWearables/biobankAccelerometerAnalysis #187: Getting temperature and light sensors recordings from AX3

LeandroCasiraghi, 2021-12-03:
Hi all. I'm starting to use the AX3 for my sleep studies and I'm glad I found this project, which I'm just starting to learn to use. I haven't been able to figure out though how to recover temperature and light information that I understand the AX3 record, which would be very useful for my research. I _think_ that I can recover some kind of temperature info if I use _deleteIntermediateFiles False_ in my accProcess call and check the resulting epoch CSV file, but I don't know if this is true. Moreover, I have no clue on where to find light levels recording. 
Is it possible to recover these using the _accelerometer_ package? Thanks.

angerhang, 2021-12-03:
@LeandroCasiraghi this is a great question. The short answer is light extraction is not currently supported in the released version. However, we do have a fork https://github.com/angerhang/biobankAccelerometerAnalysis/blob/f3ff7798524ea883206ff3cb10177bef7dc13a7c/java/AxivityReader.java#L326 that allows you to get the light information in lux.

A question we should really ask is that how much lux recording from AX3 is helpful? If you look at the hardware design, the light sensor is really hidden away. Despite the fact that we tried to make use of the light information in one of our previous studies, we decided not to use it in the end because of the lack of readability.

However, do let us know what you find out if you decide to analyse the AX3 light information in the end ;D

LeandroCasiraghi, 2021-12-03:
> @LeandroCasiraghi this is a great question. The short answer is light extraction is not currently supported in the released version. However, we do have a fork [...]

Thanks for the quick reply, @angerhang . I'll check the fork and see if I can get it to work. As for the temperature, am I right in that the way to recover the data is to keep the intermediate file and look into the _temp_ column?

As for your question on light levels, you're totally right. It's difficult to say how reliable it is, and I don't have any guesses so far as I haven't been able to recover any recorded data in this sense ha. My idea is to take some recordings along our usual light measuring instruments, see how they correlate, and decide whether it's possible to calibrate the AX3's values to get a sensible measure of light levels. But even if that is not possible, for some of my research, it would still be useful to see at least relative light-exposure patterns, even if I cannot determine the actual light levels.

angerhang, 2021-12-03:
Makes sense. Unfortunately, we don't have the capacity right now to add the lux support from our side. 

For temperature you are on the right track but @chanshing or @aidendoherty probably know better.

aidendoherty, 2021-12-14:
Nothing to add from me. @LeandroCasiraghi - I hope you have enough information to continue (in some sort of hack way) with what you would like to do?

LeandroCasiraghi, 2021-12-14:
Thanks for checking in! I tried to work with the quoted fork for pulling out the lux values but couldn't make it work. Hopefully I'll be able to work around it to get the light values soon!

closed 2021-12-14 by LeandroCasiraghi

aidendoherty, 2021-12-14:
@LeandroCasiraghi  - it might be worth having a look at https://github.com/digitalinteraction/openmovement to see if there is any potential solution. Hope this helps.