OxWearables/biobankAccelerometerAnalysis #192: Docs for collating JSON to csv

R-Walmsley, 2022-01-24:
This might be flagging a known issue, but I think the docs for collating JSON to csv don't match the available function 

- accCollateSummary on readthedocs
- collateSummary in utils.py

(I also think writeCmds might have the same issue?)

fao @chanshing 

cc @angerhang

chanshing, 2022-01-25:
Hi @R-Walmsley 
`accCollateSummary` is an entry point, so can be called directly in bash. You're right that the underlying function that it calls is `utils.collateSummary`. The entry point file is https://github.com/activityMonitoring/biobankAccelerometerAnalysis/blob/master/accelerometer/accCollateSummary.py -- all it does is to pipe the command line arguments to `utils.collateSummary`. 

I used a more explicit name for the entry point because 1) we're in bash's namespace so probably better to be more specific 2) follow the pattern for the other entry points `accProcess`, `accPlot`, ...

All entry points are defined in https://github.com/activityMonitoring/biobankAccelerometerAnalysis/blob/c05a10fbaeecde05c2839d26f2e2f5df310a81df/setup.py#L48-L54

R-Walmsley, 2022-01-25:
Oh that's clever! Thanks :)