OxWearables/biobankAccelerometerAnalysis #193: ValueError when using --epochPeriod 60

sehamsick, 2022-04-04:
Hello,

I wanted to change the epoch period to 60s (standard is 30s) and received an error. However, other period lengths which are not exactly 60s work well e.g. 50s and 120s:

So using `accProcess sample_file.cwa --epochPeriod 50` works fine, and `accProcess sample_file.cwa --epochPeriod 120` works fine as well. However, I get the following error in the "Summarizing" step using `accProcess samplefile.cwa --epochPeriod 60`:

```
> File "/home.ec2/ec2-user/miniconda3/envs/actigraphy/lib/python3.9/site-packages/accelerometer/accProcess.py", line 326, in main epochData, labels = accelerometer.summarisation.getActivitySummary(
> File "/home.ec2/ec2-user/miniconda3/envs/actigraphy/lib/python3.9/site-packages/accelerometer/summarisation.py", line 117, in getActivitySummary  data = resolveNonWear(data, stationaryStd, minNonWearDuration, summary)
> File "/home.ec2/ec2-user/miniconda3/envs/actigraphy/lib/python3.9/site-packages/accelerometer/summarisation.py", line 210, in resolveNonWear epochInDays = pd.Timedelta(pd.infer_freq(data.index)).total_seconds() / (60 * 60 * 24)
> File "pandas/_libs/tslibs/timedeltas.pyx", line 1344, in pandas._libs.tslibs.timedeltas.Timedelta.__new__
> File "pandas/_libs/tslibs/timedeltas.pyx", line 525, in pandas._libs.tslibs.timedeltas.parse_timedelta_string
> ValueError: unit abbreviation w/o a number
```

closed 2022-04-05 by chanshing

chanshing, 2022-04-05:
@sehamsick Great catch! This should be fixed now. Upgrade with `pip install -U accelerometer`

sehamsick, 2022-04-06:
@chanshing, it works now! thanks for fixing it so quickly!