OxWearables/biobankAccelerometerAnalysis #199: Axitivity timezone says `Europe/London` despite user input

angerhang, 2022-06-15:
``accProcess sampleFile --timeZone 'America/New_York`` and ``accProcess sampleFile`` produce the same timestamp in epoch and time-series files with tz being `Europe/London`. Only UNIX time in `npy` files are changed.

This is likely to be wrong. 

Not sure about what the behaviour for GeneActive. But it looks like BBAA fails for timeZone flag in general. This is a critical bug for non-UTC data.

angerhang, 2022-06-15:
I am not particularly familiar with the Axitivity reader. @chanshing would you mind working on this issue when you have the time? I will work on the GT3X time zone ticket instead.

chanshing, 2022-06-20:
> `accProcess sampleFile --timeZone 'America/New_York` and `accProcess sampleFile` produce the same timestamp in epoch and time-series files with tz being `Europe/London`. Only UNIX time in `npy` files are changed.

Hang, I am not able to reproduce this.

TimeSeries file:
[image]

Epoch file:
[image]

Are you sure you checked the right output files?

chanshing, 2022-06-20:
> This seems wrong:
> 
> https://github.com/OxWearables/biobankAccelerometerAnalysis/blob/0b8db7353c49c543de0d52369a299e79049da334/accelerometer/java/DeviceReader.java#L191
> 
> When a user supplies a time zone such as `America/New_York`, the adjusted time should observe DST directly without we having to correct it manually. A manual update is only required if we have a time zone like `EST` https://stackoverflow.com/questions/10545960/how-to-tackle-daylight-savings-using-timezone-in-java

This is correct for Axivity, but maybe not for the other devices (when I first implemented this for Axivity, I wanted to put this piece of code inside AxivityParser and not DeviceReader, because I'm aware that other devices may do things differently).

closed 2022-06-20 by chanshing