OxWearables/biobankAccelerometerAnalysis #203: GENEActive not parsing the tz/dst correctly

angerhang, 2022-06-28:
At the moment, GENEActive devices are reading local time as if they are true unix time, which is incorrect. This will create issues for daylight savings and different time zones. Something like AX3 needs to be done for GENEActive devices.

angerhang, 2022-09-26:
@chanshing do we have an estimated timeline on this on this or I will do the change myself? I need to get this done soon to start training my model?

chanshing, 2022-09-28:
> @chanshing do we have an estimated timeline on this on this or I will do the change myself? I need to get this done soon to start training my model?

Sorry Hang, just reading this. I assumed you would fix it as you're probably more familiar with the GENEActiv device than me.

angerhang, 2023-04-04:
I will have to reopen this because bba currently doesn't deal with daylight saving and time zone properly. I am having a hot fix in my code to manually move the data back by one hour. But need to raise this issue for future awareness.