OxWearables/biobankAccelerometerAnalysis #207: ArrayIndexOutOfBoundsExceptionl

johnpark12, 2022-07-28:
I'm running this with openjdk 11.0.15 2022-04-19 and Python 3.9.13

When attempting to analyze a file with accProcess test2.csv --csvTimeFormat 'MM/dd/yyyy HH:mm:ss:SSS'
I get:

`java.lang.ArrayIndexOutOfBoundsException: Index 4 out of bounds for length 4
	at CsvReader.readCSVEpochs(CsvReader.java:80)
	at AccelerometerParser.main(AccelerometerParser.java:224)
error reading/writing file test2.csv: java.lang.ArrayIndexOutOfBoundsException: Index 4 out of bounds for length 4`

This happens even when trying to run the csv demo commands. Any suggestions would be appreciated.

chanshing, 2022-07-29:
Hi @johnpark12 
Can you share header (column names) of the file? Or the first few rows...

johnpark12, 2022-07-29:
Actually I figured out what the issue was. Since I lack temperature data, I should have set the flag --csvTimeXYZTempColsIndex 0,1,2,3

closed 2022-07-29 by johnpark12