OxWearables/biobankAccelerometerAnalysis #210: how to deal with the raw data

yummyyimao, 2022-09-01:
Hey, I have received the raw accelerometer dataset from UK biobank. I wonder if  I can directly use the raw data to analyze sleep parameters using the GGIR package or firstly do the data process you mentioned in the paper: Large Scale Population Assessment of Physical Activity Using Wrist Worn Accelerometers: The UK Biobank Study?

Thanks a lot for your kindly help
yimao

chanshing, 2022-09-01:
Hi @yummyyimao I don't know the full details of GGIR and whether they process the raw Axivity files in the same way.
I believe they do, better but contact them to double-check.
The paper you mention was produced using this current tool.
Here's the usage guide in case it helps: https://biobankaccanalysis.readthedocs.io/en/latest/index.html

yummyyimao, 2022-09-02:
Thank you for your help for my questions, best wishes to you.

closed 2022-09-02 by chanshing