OxWearables/biobankAccelerometerAnalysis #222: Deprecated class in pandas 2.0

spiousas, 2023-04-20:
I freshly installed accelerometer in a new Python Venv and it installed the last version of pandas (2.0). When I tried to run a tutorial I found an error related to a deprecated class in pandas 2.0.

```
from pandas import (to_datetime, Int64Index, DatetimeIndex, Period,
ImportError: cannot import name 'Int64Index' from 'pandas' (/Users/ignacio/opt/anaconda3/lib/python3.8/site-packages/pandas/__init__.py)

```

chanshing, 2023-04-21:
Thanks for bringing this up. Can you try downgrading it: `pip install pandas==1.2.5` ?

spiousas, 2023-04-21:
Yes, I solved it by downgrading to `pandas==1.2.4` and it works perfectly, but if you follow the tutorial it installs the last pandas version by default. 

Thanks!

chanshing, 2023-05-09:
This should be fixed now the newest release. Thanks for spotting it!

closed 2023-05-09 by chanshing