OxWearables/biobankAccelerometerAnalysis #224: issue in fresh installation: TypeError

smallsr, 2023-05-03:
Getting this error when summarizing data using sample.cwa or any other file  

Traceback (most recent call last):
  File "/mnt/c/Users/ScottS/accelerometer/baa/bin/accProcess", line 8, in <module>
    sys.exit(main())
  File "/mnt/c/Users/ScottS/accelerometer/baa/lib/python3.8/site-packages/accelerometer/accProcess.py", line 326, in main
    epochData, labels = accelerometer.summarisation.getActivitySummary(
  File "/mnt/c/Users/ScottS/accelerometer/baa/lib/python3.8/site-packages/accelerometer/summarisation.py", line 117, in getActivitySummary
    data = resolveInterrupts(data, epochPeriod, summary)
  File "/mnt/c/Users/ScottS/accelerometer/baa/lib/python3.8/site-packages/accelerometer/summarisation.py", line 176, in resolveInterrupts
    data['missing'] = data.isna().any(1)
TypeError: any() takes 1 positional argument but 2 were given

smallsr, 2023-05-04:
Per @chanshing, this is due to a recent pandas recent major update.  Downgrading to pandas==1.5 fixes this error.pwd

chanshing, 2023-05-09:
https://github.com/OxWearables/biobankAccelerometerAnalysis/issues/222

closed 2023-05-09 by chanshing

Rui127, 2024-06-14:
still facing the same error using.bin file after downgrading pandas

chanshing, 2024-06-14:
@Rui127 Please can you say which version of `accelerometer` package are you using?

```bash
pip freeze | grep accelerometer
```

Rui127, 2024-06-14:
accelerometer==6.2.3

chanshing, 2024-06-15:
@Rui127 Thanks. Can you try upgrading to latest version of `accelerometer`? The TypeError issue mentioned should have been fixed in the later versions.
```bash
pip install -U accelerometer
```

Rui127, 2024-06-15:
@chanshing I upgraded to the latest version. Unfortunately I am still getting the same error below
2024-06-15 07:11:45	=== Summarizing ===
/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/accelerometer/summarisation.py:79: FutureWarning: The argument 'date_parser' is deprecated and will be removed in a future version. Please use 'date_format' instead, or read your data in as 'object' dtype and then call 'to_datetime'.
  data = pd.read_csv(epochFile, index_col=['time'], parse_dates=['time'], date_parser=utils.date_parser)
/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/accelerometer/summarisation.py:169: FutureWarning: 'S' is deprecated and will be removed in a future version. Please use 's' instead of 'S'.
  epochPeriod = pd.Timedelta(epochPeriod, unit='S')
Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.9/bin/accProcess", line 8, in <module>
    sys.exit(main())
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/accelerometer/accProcess.py", line 326, in main
    epochData, labels = accelerometer.summarisation.getActivitySummary(
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/accelerometer/summarisation.py", line 117, in getActivitySummary
    data = resolveInterrupts(data, epochPeriod, summary)
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/accelerometer/summarisation.py", line 176, in resolveInterrupts
    data['missing'] = data.isna().any(1)
TypeError: any() takes 1 positional argument but 2 were given

chanshing, 2024-06-15:
@Rui127 Please can you also share pandas version? `pip freeze | grep pandas`

Rui127, 2024-06-15:
@chanshing 
pandas==1.3.5
pandas-flavor==0.6.0

chanshing, 2024-06-15:
`pandas-flavor` is not in our package's dependency list so I am curious how it got there. Can you try uninstalling it and see if it works after that?
`pip uninstall pandas-flavor`

chanshing, 2024-06-15:
@Rui127 Also have you installed `accelerometer` in a separate environment?
https://github.com/OxWearables/biobankAccelerometerAnalysis?tab=readme-ov-file#install

Rui127, 2024-06-15:
@chanshing Thanks for pointing it out. I activate environment however it did not solve the problem. 
accelerometer==7.1.1
pandas==1.3.5

Rui127, 2024-06-15:
@chanshing Thank you for helping and please let me know if you can download it
https://drive.google.com/file/d/1PFJRLqMiB2f7Qt6u45JhEujq7TbdeE2E/view?usp=drive_link

chanshing, 2024-06-15:
@Rui127 It doesn't seem to work - restricted access

chanshing, 2024-06-16:
[test-summary.json](https://github.com/user-attachments/files/15856181/test-summary.json)
[image]

@Rui127 I was able to process your file just fine. Maybe it's something wrong with the installation. You could try reinstalling:
```bash
conda deactivate  # exit current environment
conda env remove -n accelerometer  # delete 'accelerometer' environment
```
then retry these steps: https://github.com/OxWearables/biobankAccelerometerAnalysis?tab=readme-ov-file#install

Can you also try `pip freeze` and paste here the list of packages installed?

Rui127, 2024-06-16:
@chanshing I realized that the environment was built under Anaconda that i previously installed. I removed Anaconda and it works now. Sorry that I did not check it and thanks for the help

chanshing, 2024-06-16:
Ok great! We do recommend anaconda for new Python users for env management, but glad you solved your problem!

Rui127, 2024-06-19:
@chanshing I am checking the test-timeSeries.csv.gz file and wondering if there's a way to extract the  light sensor data in the bin file as well e.g., mean or max lux of 30s bins

chanshing, 2024-06-19:
> @chanshing I am checking the test-timeSeries.csv.gz file and wondering if there's a way to extract the light sensor data in the bin file as well e.g., mean or max lux of 30s bins

Unfortunately we don't think this will be supported in the near future, but will let you know if things change!

aidendoherty, 2024-06-20:
@Rui127 - just a note to say that we will of course welcome a pull request if you would like to try including this feature.