OxWearables/biobankAccelerometerAnalysis #227: Misaligned address error on MacOS

angerhang, 2023-07-10:
@chanshing so Andrew received this misaligned address error on BBA which I couldn't reproduce on my Mac machine. Would you mind taking a look?

```
> python --version               
Python 3.10.11
 
> java -version                 
openjdk version "20.0.1" 2023-04-18
OpenJDK Runtime Environment Temurin-20.0.1+9 (build 20.0.1+9)
OpenJDK 64-Bit Server VM Temurin-20.0.1+9 (build 20.0.1+9, mixed mode)
 
MacOS 13.4.1
 

 
 
> wget -P data/ http://gas.ndph.ox.ac.uk/aidend/accModels/sample.cwa.gz
URL transformed to HTTPS due to an HSTS policy
--2023-07-10 14:59:57--  https://gas.ndph.ox.ac.uk/aidend/accModels/sample.cwa.gz
Resolving [gas.ndph.ox.ac.uk](http://gas.ndph.ox.ac.uk/) ([gas.ndph.ox.ac.uk](http://gas.ndph.ox.ac.uk/))... 163.1.206.75
Connecting to [gas.ndph.ox.ac.uk](http://gas.ndph.ox.ac.uk/) ([gas.ndph.ox.ac.uk](http://gas.ndph.ox.ac.uk/))|163.1.206.75|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 72765789 (69M) [application/x-gzip]
Saving to: ‘data/sample.cwa.gz’
 
sample.cwa.gz                         100%[======================================================================>]  69.39M  9.21MB/s    in 7.4s
 
2023-07-10 15:00:05 (9.32 MB/s) - ‘data/sample.cwa.gz’ saved [72765789/72765789]
 
> accProcess data/sample.cwa.gz                                                                         
 
Processing file 'data/sample.cwa.gz' with these arguments:
 
activityClassification    : True
activityModel             : walmsley
calOffset                 : [0.0, 0.0, 0.0]
calSlope                  : [1.0, 1.0, 1.0]
calTemp                   : [0.0, 0.0, 0.0]
calibrationSphereCriteria : 0.3
csvSampleRate             : None
csvStartRow               : 1
csvStartTime              : None
csvTimeFormat             : yyyy-MM-dd HH:mm:ss.SSSxxxx '['VV']'
csvTimeXYZTempColsIndex   : 0,1,2,3,4
deleteIntermediateFiles   : True
endTime                   : None
epochFile                 : /Users/180341c/src/test_baa/data/sample-epoch.csv.gz
epochPeriod               : 30
extractFeatures           : True
fourierFrequency          : False
fourierWithAcc            : False
inputFile                 : data/sample.cwa.gz
intensityDistribution     : False
m10l5                     : False
meanTemp                  : None
mgCutPointMVPA            : 100
mgCutPointVPA             : 425
nonWearFile               : /Users/180341c/src/test_baa/data/sample-nonWearBouts.csv.gz
npyFile                   : /Users/180341c/src/test_baa/data/sample.npy
npyOutput                 : False
outputFolder              : /Users/180341c/src/test_baa/data
processInputFile          : True
psd                       : False
rawDataParser             : AccelerometerParser
rawFile                   : /Users/180341c/src/test_baa/data/sample.csv.gz
rawOutput                 : False
resampleMethod            : linear
sampleRate                : 100
skipCalibration           : False
startTime                 : None
stationaryFile            : /Users/180341c/src/test_baa/data/sample-stationaryPoints.csv.gz
stationaryStd             : 13
summaryFile               : /Users/180341c/src/test_baa/data/sample-summary.json
timeSeriesDateColumn      : False
timeShift                 : 0
timeZone                  : Europe/London
tsFile                    : /Users/180341c/src/test_baa/data/sample-timeSeries.csv.gz
useFilter                 : True
verbose                   : False
 
2023-07-10 15:00:11   === Calibrating ===
Intermediate file: /Users/180341c/src/test_baa/data/sample-stationaryPoints.csv.gz
Session start: 2014-05-07T13:29:51.689300537+01:00[Europe/London]
Interrupt of length: 86.575s, at epoch 2014-05-07T16:28:00.439+01:00[Europe/London]
from: 2014-05-07T16:28:02.912+01:00[Europe/London]
to  : 2014-05-07T16:29:29.487+01:00[Europe/London]
fish: Job 1, 'accProcess data/sample.cwa.gz' terminated by signal SIGBUS (Misaligned address error)
```

chanshing, 2023-07-10:
@angerhang Have you tried it on your M1 machine? What machine does Andrew have?
Can Andrew try `openjdk=8`?
`$ conda install openjdk=8`

angerhang, 2023-07-10:
> @angerhang Have you tried it on your M1 machine? What machine does Andrew have?
> Can Andrew try `openjdk=8`?
> `$ conda install openjdk=8`

Andrew has an M1 machine and me too. Yes I don't think he has tried jdk=8 yet which we probably should do :D Thanks for the quick response Shing!

chanshing, 2023-07-15:
It seemed it was an issue with Java that got fixed now.

closed 2023-07-15 by chanshing