OxWearables/biobankAccelerometerAnalysis #235: There are some trouble when dealing with the raw file of the UK biobank Accelerometer by conda prompt

36310489, 2023-10-27:
There are some trouble when dealing with the raw file of the UK biobank Accelerometer by conda prompt. Could anyone offer help?
Here is the code:
(accelerometer) C:\Users\86137>accProcess  C/Users/86137/ukbcwa.cwa/
Processing file 'C/Users/86137/ukbcwa.cwa/' with these arguments:

activityClassification    : True
activityModel             : walmsley
calOffset                 : [0.0, 0.0, 0.0]
calSlope                  : [1.0, 1.0, 1.0]
calTemp                   : [0.0, 0.0, 0.0]
calibrationSphereCriteria : 0.3
csvSampleRate             : None
csvStartRow               : 1
csvStartTime              : None
csvTimeFormat             : yyyy-MM-dd HH:mm:ss.SSSxxxx '['VV']'
csvTimeXYZTempColsIndex   : 0,1,2,3,4
deleteIntermediateFiles   : True
endTime                   : None
epochFile                 : C:\Users\86137\C\Users\86137\ukbcwa.cwa\-epoch.csv.gz
epochPeriod               : 30
extractFeatures           : True
fourierFrequency          : False
fourierWithAcc            : False
inputFile                 : C/Users/86137/ukbcwa.cwa/
intensityDistribution     : False
m10l5                     : False
meanTemp                  : None
mgCutPointMVPA            : 100
mgCutPointVPA             : 425
nonWearFile               : C:\Users\86137\C\Users\86137\ukbcwa.cwa\-nonWearBouts.csv.gz
npyFile                   : C:\Users\86137\C\Users\86137\ukbcwa.cwa\.npy
npyOutput                 : False
outputFolder              : C:\Users\86137\C\Users\86137\ukbcwa.cwa
processInputFile          : True
psd                       : False
rawDataParser             : AccelerometerParser
rawFile                   : C:\Users\86137\C\Users\86137\ukbcwa.cwa\.csv.gz
rawOutput                 : False
resampleMethod            : linear
sampleRate                : 100
skipCalibration           : False
startTime                 : None
stationaryFile            : C:\Users\86137\C\Users\86137\ukbcwa.cwa\-stationaryPoints.csv.gz
stationaryStd             : 13
summaryFile               : C:\Users\86137\C\Users\86137\ukbcwa.cwa\-summary.json
timeSeriesDateColumn      : False
timeShift                 : 0
timeZone                  : Europe/London
tsFile                    : C:\Users\86137\C\Users\86137\ukbcwa.cwa\-timeSeries.csv.gz
useFilter                 : True
verbose                   : False
ERROR: Cannot get deviceId for file: C/Users/86137/ukbcwa.cwa/

2023-10-27 21:27:11     === Calibrating ===
Intermediate file: C:\Users\86137\C\Users\86137\ukbcwa.cwa\-stationaryPoints.csv.gz
Unrecognised file format for: C/Users/86137/ukbcwa.cwa/
['java', '-classpath', 'F:\\ana\\envs\\accelerometer\\lib\\site-packages\\accelerometer/java/;F:\\ana\\envs\\accelerometer\\lib\\site-packages\\accelerometer/java/JTransforms-3.1-with-dependencies.jar', '-XX:ParallelGCThreads=1', 'AccelerometerParser', 'C/Users/86137/ukbcwa.cwa/', 'timeZone:Europe/London', 'timeShift:0', 'outputFile:C:\\Users\\86137\\C\\Users\\86137\\ukbcwa.cwa\\-stationaryPoints.csv.gz', 'verbose:False', 'filter:True', 'getStationaryBouts:true', 'epochPeriod:10', 'stationaryStd:0.013', 'sampleRate:100', "csvTimeFormat:yyyy-MM-dd HH:mm:ss.SSSxxxx '['VV']'", 'csvStartRow:1', 'csvTimeXYZTempColsIndex:0,1,2,3,4']
Error: java calibration failed, exit  4294967295

36310489, 2023-10-29:
I'm using the Windows system.

chanshing, 2023-10-30:
Hi @36310489 

Does it work with the sample file provided here? http://gas.ndph.ox.ac.uk/aidend/accModels/sample.cwa.gz

36310489, 2023-10-31:
I'm using python3.9 and Java 1.8.0 now. I can get the outcome with the sample, but there are something wrong with  .cwa file of the UK biobank.
Here is the code:
C:\Users\86137>accProcess C:/Users/86137/ukbcwa.cwa
Processing file 'C:/Users/86137/ukbcwa.cwa' with these arguments:

activityClassification    : True
activityModel             : walmsley
calOffset                 : [0.0, 0.0, 0.0]
calSlope                  : [1.0, 1.0, 1.0]
calTemp                   : [0.0, 0.0, 0.0]
calibrationSphereCriteria : 0.3
csvSampleRate             : None
csvStartRow               : 1
csvStartTime              : None
csvTimeFormat             : yyyy-MM-dd HH:mm:ss.SSSxxxx '['VV']'
csvTimeXYZTempColsIndex   : 0,1,2,3,4
deleteIntermediateFiles   : True
endTime                   : None
epochFile                 : C:\Users\86137\ukbcwa-epoch.csv.gz
epochPeriod               : 30
extractFeatures           : True
fourierFrequency          : False
fourierWithAcc            : False
inputFile                 : C:/Users/86137/ukbcwa.cwa
intensityDistribution     : False
m10l5                     : False
meanTemp                  : None
mgCutPointMVPA            : 100
mgCutPointVPA             : 425
nonWearFile               : C:\Users\86137\ukbcwa-nonWearBouts.csv.gz
npyFile                   : C:\Users\86137\ukbcwa.npy
npyOutput                 : False
outputFolder              : C:\Users\86137
processInputFile          : True
psd                       : False
rawDataParser             : AccelerometerParser
rawFile                   : C:\Users\86137\ukbcwa.csv.gz
rawOutput                 : False
resampleMethod            : linear
sampleRate                : 100
skipCalibration           : False
startTime                 : None
stationaryFile            : C:\Users\86137\ukbcwa-stationaryPoints.csv.gz
stationaryStd             : 13
summaryFile               : C:\Users\86137\ukbcwa-summary.json
timeSeriesDateColumn      : False
timeShift                 : 0
timeZone                  : Europe/London
tsFile                    : C:\Users\86137\ukbcwa-timeSeries.csv.gz
useFilter                 : True
verbose                   : False
ERROR: in getDeviceId("C:/Users/86137/ukbcwa.cwa")
A deviceId value could not be found in input file header,
         this usually occurs when the file is not an Axivity .cwa accelerometer
         file. Exiting...

C:\Users\86137>accProcess C:/Users/86137/Downloads/sample.cwa.gz
Processing file 'C:/Users/86137/Downloads/sample.cwa.gz' with these arguments:

activityClassification    : True
activityModel             : walmsley
calOffset                 : [0.0, 0.0, 0.0]
calSlope                  : [1.0, 1.0, 1.0]
calTemp                   : [0.0, 0.0, 0.0]
calibrationSphereCriteria : 0.3
csvSampleRate             : None
csvStartRow               : 1
csvStartTime              : None
csvTimeFormat             : yyyy-MM-dd HH:mm:ss.SSSxxxx '['VV']'
csvTimeXYZTempColsIndex   : 0,1,2,3,4
deleteIntermediateFiles   : True
endTime                   : None
epochFile                 : C:\Users\86137\Downloads\sample-epoch.csv.gz
epochPeriod               : 30
extractFeatures           : True
fourierFrequency          : False
fourierWithAcc            : False
inputFile                 : C:/Users/86137/Downloads/sample.cwa.gz
intensityDistribution     : False
m10l5                     : False
meanTemp                  : None
mgCutPointMVPA            : 100
mgCutPointVPA             : 425
nonWearFile               : C:\Users\86137\Downloads\sample-nonWearBouts.csv.gz
npyFile                   : C:\Users\86137\Downloads\sample.npy
npyOutput                 : False
outputFolder              : C:\Users\86137\Downloads
processInputFile          : True
psd                       : False
rawDataParser             : AccelerometerParser
rawFile                   : C:\Users\86137\Downloads\sample.csv.gz
rawOutput                 : False
resampleMethod            : linear
sampleRate                : 100
skipCalibration           : False
startTime                 : None
stationaryFile            : C:\Users\86137\Downloads\sample-stationaryPoints.csv.gz
stationaryStd             : 13
summaryFile               : C:\Users\86137\Downloads\sample-summary.json
timeSeriesDateColumn      : False
timeShift                 : 0
timeZone                  : Europe/London
tsFile                    : C:\Users\86137\Downloads\sample-timeSeries.csv.gz
useFilter                 : True
verbose                   : False

2023-10-31 21:09:25     === Calibrating ===
Intermediate file: C:\Users\86137\Downloads\sample-stationaryPoints.csv.gz
Session start: 2014-05-07T13:29:51.689300537+01:00[Europe/London]
Interrupt of length: 86.575s, at epoch 2014-05-07T16:28:00.439+01:00[Europe/London]
 from: 2014-05-07T16:28:02.912+01:00[Europe/London]
 to  : 2014-05-07T16:29:29.487+01:00[Europe/London]

2023-10-31 21:09:52     === Extracting features ===
Intermediate file: C:\Users\86137\Downloads\sample-epoch.csv.gz
Session start: 2014-05-07T13:29:51.689300537+01:00[Europe/London]
Interrupt of length: 86.575s, at epoch 2014-05-07T16:27:50.439+01:00[Europe/London]
 from: 2014-05-07T16:28:02.912+01:00[Europe/London]
 to  : 2014-05-07T16:29:29.487+01:00[Europe/London]

2023-10-31 21:11:32     === Summarizing ===
Downloading https://wearables-files.ndph.ox.ac.uk/files/models/biobankAccelerometerAnalysis/walmsley/20220210_sk_1_0_2.tar...
125 rows with NaN or Inf values, out of 16841

2023-10-31 21:12:18     === Short summary ===
{
    "file-name": "C:/Users/86137/Downloads/sample.cwa.gz",
    "file-startTime": "2014-05-07 13:29:50.439000+0100 [Europe/London]",
    "file-endTime": "2014-05-13 09:49:50.439000+0100 [Europe/London]",
    "acc-overall-avg": 32.83246363816311,
    "wearTime-overall(days)": 5.804166666666667,
    "nonWearTime-overall(days)": 0.043402777777777776,
    "quality-goodWearTime": 1
}
Full summary written to: C:\Users\86137\Downloads\sample-summary.json

2023-10-31 21:12:18     In total, processing took 173.571452 seconds

chanshing, 2023-10-31:
I suspect the ukbcwa file is corrupted. Does it only occur with this particular file? Do you have another file you can try with?

36310489, 2023-10-31:
I think so because I can get the outcome too with the sample datafile of the UK biobank accelerometer. Maybe there are something wrong with the raw ukbcwa file. Thank you for your help!

closed 2024-02-09 by chanshing