OxWearables/biobankAccelerometerAnalysis #252: bug

Bianlab2023, 2024-04-17:
Error: java calibration failed, exit  254

Bianlab2023, 2024-04-17:
error reading/writing file data/sample-small.csv: java.lang.ArrayIndexOutOfBoundsException: 4
['java', '-classpath', '/share/home/zlbian/.conda/envs/accelerometer/lib/python3.9/site-packages/accelerometer/java/:/share/home/zlbian/.conda/envs/accelerometer/lib/python3.9/site-packages/accelerometer/java/JTransforms-3.1-with-dependencies.jar', '-XX:ParallelGCThreads=1', 'AccelerometerParser', 'data/sample-small.csv', 'timeZone:Europe/London', 'timeShift:0', 'outputFile:/share2/home/zlbian/NHANES/data/sample-small-stationaryPoints.csv.gz', 'verbose:False', 'filter:True', 'getStationaryBouts:true', 'epochPeriod:10', 'stationaryStd:0.013', 'sampleRate:100', "csvTimeFormat:yyyy-MM-dd HH:mm:ss.SSSxxxx '['VV']'", 'csvStartRow:1', 'csvTimeXYZTempColsIndex:0,1,2,3,4']
Error: java calibration failed, exit 254

Bianlab2023, 2024-04-17:
it does not work with sample file .

chanshing, 2024-04-17:
Hi @Bianlab2023 Could you share the sample file you are using?

Bianlab2023, 2024-04-17:
i used the sample file  that you provided.

---- Replied Message ----
| From | Shing ***@***.***> |
| Date | 04/18/2024 01:03 |
| To | OxWearables/biobankAccelerometerAnalysis ***@***.***> |
| Cc | Bianlab2023 ***@***.***>,
Mention ***@***.***> |
| Subject | Re: [OxWearables/biobankAccelerometerAnalysis] bug (Issue #252) |

Hi @Bianlab2023 Could you share the sample file you are using?

Bianlab2023, 2024-04-18:
"http://gas.ndph.ox.ac.uk/aidend/accModels/sample-small.csv.gz"

closed 2024-04-18 by chanshing