OxWearables/biobankAccelerometerAnalysis #258: can't accProcess .csv directly

tbonewmy, 2024-08-19:
Hi there,
I tried to use accProcess to process csv file directly as this link suggested https://biobankaccanalysis.readthedocs.io/en/latest/usage.html#basic-usage.
However, when I give either a csv file or a csv.gz file, the call() function in device.py give a non-zero exit code. The process then fail and exist.

the command line I use is `(myenv) PS C:\Users\my> accProcess SomeDiractory/sample-small.csv.gz`
the csv data is downloaded from the place mentioned in above link in the format of time,x,y,z.
My machine is windows.
Python 3.9
accelerometer 7.1.2

I feel like it's the csv format doesn't compatible with the java code, since everything seems normal until call the java functions.
Please help me.

chanshing, 2024-08-19:
Hi @tbonewmy Can you try unzipping it first?

tbonewmy, 2024-08-19:
> Hi @tbonewmy Can you try unzipping it first?

Yes, I tried both zipped and upzipped files. sample-small.csv.gz and sample-small.csv. Neither of them works. Below is the error message from the command line end.
[image]

And as I am writing, I think I found the cause to my problem. I tried it. It works. The issue is the example file only have time,x,y,z columns. But the default csvTimeXYZTempColsIndex parameter has values 0,1,2,3,4. So the indices don't match. I gave it 0,1,2,3 and problem solved.

chanshing, 2024-08-19:
Thanks @tbonewmy I'll fix the documentation soon

closed 2024-09-10 by tbonewmy